UZ-SLAMLab/ORB_SLAM3 #828: Segmentation fault (core dumped)

IAMHAADICOOL, 2023-11-08:
./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/RealSense_D435i.yaml
I ran the above command in the root ORB_SLAM3 folder and following is the output I get:
Sensor supports the following options:

  0: Backlight Compensation is not supported
  1: Brightness is not supported
  2: Contrast is not supported
  3: Exposure
       Description   : Depth Exposure (usec)
       Current Value : 8500
  4: Gain
       Description   : UVC image gain
       Current Value : 16
  5: Gamma is not supported
  6: Hue is not supported
  7: Saturation is not supported
  8: Sharpness is not supported
  9: White Balance is not supported
  10: Enable Auto Exposure
       Description   : Enable Auto Exposure
       Current Value : 1
  11: Enable Auto White Balance is not supported
  12: Visual Preset
       Description   : Advanced-Mode Preset
       Current Value : 0
  13: Laser Power
       Description   : Manual laser power in mw. applicable only when laser power mode is set to Manual
       Current Value : 150
  14: Accuracy is not supported
  15: Motion Range is not supported
  16: Filter Option is not supported
  17: Confidence Threshold is not supported
  18: Emitter Enabled
       Description   : Emitter select, 0-disable all emitters, 1-enable laser, 2-enable laser auto (opt), 3-enable LED (opt)
       Current Value : 0
  19: Frames Queue Size
       Description   : Max number of frames you can hold at a given time. Increasing this number will reduce frame drops but increase latency, and vice versa
       Current Value : 16
  20: Total Frame Drops is not supported
  21: Fisheye Auto Exposure Mode is not supported
  22: Power Line Frequency is not supported
  23: Asic Temperature is not supported
  24: Error Polling Enabled
       Description   : Enable / disable polling of camera internal errors
       Current Value : 1
  25: Projector Temperature is not supported
  26: Output Trigger Enabled
       Description   : Generate trigger from the camera to external device once per frame
       Current Value : 0
  27: Motion Module Temperature is not supported
  28: Depth Units
       Description   : Number of meters represented by a single depth unit
       Current Value : 0.001
  29: Enable Motion Correction is not supported
  30: Auto Exposure Priority is not supported
  31: Color Scheme is not supported
  32: Histogram Equalization Enabled is not supported
  33: Min Distance is not supported
  34: Max Distance is not supported
  35: Texture Source is not supported
  36: Filter Magnitude is not supported
  37: Filter Smooth Alpha is not supported
  38: Filter Smooth Delta is not supported
  39: Holes Fill is not supported
  40: Stereo Baseline
       Description   : Distance in mm between the stereo imagers
       Current Value : 49.7533
  41: Auto Exposure Converge Step is not supported
  42: Inter Cam Sync Mode
       Description   : Inter-camera synchronization mode: 0:Default, 1:Master, 2:Slave, 3:Full Salve, 4-258:Genlock with burst count of 1-255 frames for each trigger, 259 and 260 for two frames per trigger with laser ON-OFF and OFF-ON.
[... 329 lines cut ...]
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo-Inertial
Loading settings from ./Examples/Stereo-Inertial/RealSense_D435i.yaml
	-Loaded camera 1
	-Loaded camera 2
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
	-Loaded IMU calibration
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 382.613 382.613 320.183 236.455 ]
Segmentation fault (core dumped)

How do I fix this?

IAMHAADICOOL, 2023-11-08:
It's a request to the developer's to kindly help resolve this issue of mine.

unicornuniform, 2023-11-11:
Start with gdb, run and then backtrace.

IAMHAADICOOL, 2023-11-11:
Sorry, but I don't get what you are saying, can you please elaborate. By the way, thank you for your response?

IAMHAADICOOL, 2023-11-11:
[image]
This is the output

IAMHAADICOOL, 2023-11-11:
Yes, I gave the arguments with the run command

IAMHAADICOOL, 2023-11-11:
[image]

IAMHAADICOOL, 2023-11-11:
[image]

IAMHAADICOOL, 2023-11-12:
Hey, I updated on the issue.
Can you please look into it?
I would be highly grateful if you could help me out with this issue.

IAMHAADICOOL, 2023-11-15:
Hey, can you please help?

IAMHAADICOOL, 2023-11-16:
Sorry, I closed it by mistake.
Hey @unicornuniform can you please help?

unicornuniform, 2023-11-16:
There could be some mistake in your yaml file. I would try running it again with the default one and see what happens.

I would also try experimenting with compiling and running one of the alternative repos. Right now I'm using https://github.com/aPR0T0/ORB_SLAM3

 https://github.com/dz306271098/ORB_SLAM3 https://github.com/viamrobotics/ORB_SLAM3 https://github.com/thien94/ORB_SLAM3 https://github.com/Soldann/MORB_SLAM https://github.com/ido7746/ORB_SLAM3 https://github.com/Mauhing/ORB_SLAM3
https://github.com/neo-knight-td/ORB_SLAM3/tree/save_load_branch

IAMHAADICOOL, 2023-11-17:
Please let me know the results as soon as possible, I request you sincerely.

unicornuniform, 2023-11-17:
I can't do any of it for you, you'll have to experiment yourself

IAMHAADICOOL, 2023-11-17:
What's the default yaml file?

IAMHAADICOOL, 2023-11-17:
I tried this one:
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/Settings.cc#L532

And now I am getting error:
[image]

This compilter appears to be too old to be supported by Eigen

IAMHAADICOOL, 2023-11-22:
> The output shows it is crashing around here, https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/src/Settings.cc#L532

I tried this one and I am getting the following error.
[image]

Abhicoder1999, 2023-11-26:
I am facing the same issue. There seems to be some issue with the "<<" operator overloading in Settings.cc  @unicornuniform https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/Settings.cc#L529

@IAMHAADICOOL In the meantime, you can comment this line, and the program should be running.
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/System.cc#L84 

I hope it helps!

IAMHAADICOOL, 2023-11-26:
Okay,I'll try your solution and will get back to you.

IAMHAADICOOL, 2023-12-15:
> according to @Abhicoder1999 comment, I found the matter is this line
> 
> ```
> for(size_t i=0; i<settings.originalCalib2_->size();i++)
> ```
> 
> `settings.originalCalib2_->size()` cannot return 0, so this issue became. but in ORB_SLAM3/include/CameraModels/GeometricCamera.h , the way to init `std::vector<float> mvParameters` was right, so the `mvParameters.size()` should be 0. It's wired.
> 
> I finally added like this in settings.cc, it's worked for me.
> 
> ```
> if (settings.sensor_ != System::IMU_STEREO){
>     for(size_t i=0; i<settings.originalCalib2_->size();i++){
>         ...
>     }
> }
> ```

Can you please specify the file name?

IAMHAADICOOL, 2023-12-15:
Still not working, tried both of your solutions, not it's just printing arbitrary numbers non stop

IAMHAADICOOL, 2023-12-16:
I lowered the library version to 2.50 and also lowered the firmware version which corresponds to library version 2.50 and the stereo ones worked. But the stereo intertial ones are still not working. It's showing "not enough accleration" in the terminal.
Any solutions?

wanqingYuan, 2023-12-18:
is your camera motionless? maybe you can move your camera faster

IAMHAADICOOL, 2023-12-18:
I moved the camera, I got some frames, but it just resets again and starts building point-cloud again. There is some desynchronization between the accelerometer and gyroscope output frequency as a similar error was faced by someone but with the T series of Realsense. I am leaving the link down of that. If you understand that solution, can you please suggest the corresponding changes for me too?

jisonZ, 2024-03-09:
@IAMHAADICOOL experiencing the same thing. Any luck so far?

atesahmet0, 2024-10-27:
Issue still persists.

15101051, 2025-06-13:
> I am facing the same issue. There seems to be some issue with the "<<" operator overloading in Settings.cc [@unicornuniform](https://github.com/unicornuniform)
> 
> [ORB_SLAM3/src/Settings.cc](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/Settings.cc#L529)
> 
> Line 529 in [4452a3c](/UZ-SLAMLab/ORB_SLAM3/commit/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4)
> 
>  ostream &operator<<(std::ostream& output, const Settings& settings){ 
> [@IAMHAADICOOL](https://github.com/IAMHAADICOOL) In the meantime, you can comment this line, and the program should be running.
> 
> [ORB_SLAM3/src/System.cc](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/System.cc#L84)
> 
> Line 84 in [4452a3c](/UZ-SLAMLab/ORB_SLAM3/commit/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4)
> 
>  cout << (*settings_) << endl; 
> I hope it helps!

This worked for me. Maybe its because `settings.sensor_` is private?