UZ-SLAMLab/ORB_SLAM3 #841: Segmentation fault inside KeyFrame::GetImuPosition()

RajS999, 2023-11-23:
I was facing `Segmentation fault` while running ORB-Slam 3 integrated with my custom application. I tried to write core dump and check the back trace. From frame 5 of stack trace below, the issue seems to be in `KeyFrame::GetImuPosition()`. Also it seem that it is trying to access pointer that is null `entry=0x0`.

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  __GI___pthread_mutex_lock (mutex=0x1100) at ../nptl/pthread_mutex_lock.c:67
67	../nptl/pthread_mutex_lock.c: No such file or directory.
[Current thread is 1 (Thread 0x7f5c317fc700 (LWP 1926529))]
(gdb) bt
#0  __GI___pthread_mutex_lock (mutex=0x1100) at ../nptl/pthread_mutex_lock.c:67
#1  0x00007f5ce49502df in __gthread_mutex_lock (__mutex=0x1100) at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749
#2  std::mutex::lock() (this=0x1100) at /usr/include/c++/9/bits/std_mutex.h:100
#3  std::unique_lock<std::mutex>::lock() (this=<synthetic pointer>, this=<synthetic pointer>) at /usr/include/c++/9/bits/unique_lock.h:141
#4  std::unique_lock<std::mutex>::unique_lock(std::mutex&) (__m=..., this=<synthetic pointer>) at /usr/include/c++/9/bits/unique_lock.h:71
#5  ORB_SLAM3::KeyFrame::GetImuPosition() (this=this@entry=0x0) at /home/my-user/workspace/my-project/ORBSlam3/src/KeyFrame.cc:150
#6  0x00007f5ce4a56e44 in ORB_SLAM3::ImuCamPose::ImuCamPose(ORB_SLAM3::KeyFrame*) (this=0x7f5c317b0380, pKF=0x0) at /home/my-user/workspace/my-project/ORBSlam3/src/G2oTypes.cc:28
#7  0x00007f5ce49a559a in ORB_SLAM3::VertexPose::VertexPose(ORB_SLAM3::KeyFrame*) (this=0x7f5bec463c40, pKF=0x0) at /home/my-user/workspace/my-project/ORBSlam3/include/G2oTypes.h:136
#8  0x00007f5ce499140e in ORB_SLAM3::Optimizer::PoseInertialOptimizationLastKeyFrame(ORB_SLAM3::Frame*, bool) (pFrame=pFrame@entry=0x7f5c66729240, bRecInit=bRecInit@entry=false)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:212
#9  0x00007f5ce48fcb09 in ORB_SLAM3::Tracking::TrackLocalMap() (this=0x7f5c66729230) at /home/my-user/workspace/my-project/ORBSlam3/src/Tracking.cc:3155
#10 0x00007f5ce49027f0 in ORB_SLAM3::Tracking::Track() (this=0x7f5c66729230) at /home/my-user/workspace/my-project/ORBSlam3/src/Tracking.cc:2292
... project specific stack frames ..
```

What could be the reason here?

PS: The application and SLAM tracking works fine for a couple of minutes processing 1000+ frames (as I can see in logs). It then fails with segmentation fault.

GLtangzero, 2024-07-02:
@RajS999 
I have the same problem, have you finished it?

EmmanuelMess, 2024-08-22:
Here:
```
#5  ORB_SLAM3::KeyFrame::GetImuPosition() (this=this@entry=0x0) at /home/my-user/workspace/my-project/ORBSlam3/src/KeyFrame.cc:150
#6  0x00007f5ce4a56e44 in ORB_SLAM3::ImuCamPose::ImuCamPose(ORB_SLAM3::KeyFrame*) (this=0x7f5c317b0380, pKF=0x0) at /home/my-user/workspace/my-project/ORBSlam3/src/G2oTypes.cc:28
#7  0x00007f5ce49a559a in ORB_SLAM3::VertexPose::VertexPose(ORB_SLAM3::KeyFrame*) (this=0x7f5bec463c40, pKF=0x0) at /home/my-user/workspace/my-project/ORBSlam3/include/G2oTypes.h:136
#8  0x00007f5ce499140e in ORB_SLAM3::Optimizer::PoseInertialOptimizationLastKeyFrame(ORB_SLAM3::Frame*, bool) (pFrame=pFrame@entry=0x7f5c66729240, bRecInit=bRecInit@entry=false)
    at /usr/include/eigen3/Eigen/src/Core/util/Memory.h:212
```

Here it implicitly converts Frame to KeyFrame.

EmmanuelMess, 2024-08-23:
Check that the IMU messages are consistent, I had issues with delayed messages, and crashes.