UZ-SLAMLab/ORB_SLAM3 #854: Black screen for a while and then starting....

Sadegh-Kalami, 2024-01-10:
Hello,

I appreciate your project and love it. However, I'm having some trouble with the algorithm when I test it with my drone dataset. Specifically, for the first 800 images, ORBSLAM doesn't detect any movement and I get a black screen (waiting for Images). I've already tried changing the ORBextractor.nFeature, ORBextractor.scaleFactor, ORBextractor.nLevels, ORBextractor.iniThFAST, and ORBextractor.minThFAST values, but I still can't get it to work. 

The Orbslam3 starts working on frame 1800, and I'm not sure why it's not working for the first 800 frames, especially since the scenario and movement in my drone dataset are all the same (same altitude and same speed). I'm stuck and I've tried all possible ways to fix it.

I would appreciate your help. If you have any ideas or suggestions, please let me know. 

Best regards and love,
Kalami

Styazoua, 2024-01-17:
I am having the same issue since I started to use ORB_SLAM3.

I noticed when I start the measurement while camera is already moving it doesn't take that long to detect any movement.
And I also noticed an improvement in initial detection while running ORB_SLAM3 in a real time application.
In both cases I had still some lost frames but it took ~2s - 5s to initialize.

I haven't tested with a lower image resolution. Maybe this could reduce the processing time.

Let me know if you can solve that problem or reduce the "waiting" time.

Best regards

Sadegh-Kalami, 2024-01-17:
in my case, I lose 1800 frames and then start detecting.
I have tested opencv ORB2 and other algorithms. The feature extractor doesn't work properly in my case.

Styazoua, 2024-01-17:
Have you also tested with the sample datasets like EuRoC, KITTI, etc... ?

Sadegh-Kalami, 2024-01-17:
yes, I have tested with all datasets, and worked well.

closed 2024-04-27 by Sadegh-Kalami

1fishingmaster, 2025-04-11:
I probably lost more than 9000 frames, how did you solve it?