UZ-SLAMLab/ORB_SLAM3 #856: Move current frame window to new position with cv::moveWindow not working.

evertale888, 2024-01-14:
Hello guys.
I try to move the current frame window/viewer to new position during starting of the orbslam3. Im using  cv::moveWindow  for that & unfortunately it does nothing. Seem like i cant change the window position. Any helps are appreciated.

AlfreddGco, 2024-01-21:
I think you should maybe try to look for the initialization of the window and see if you are able to set an initial position in there. Not start the window and then try to move it, but start it where you want it to be

evertale888, 2024-01-21:
> I think you should maybe try to look for the initialization of the window and see if you are able to set an initial position in there. Not start the window and then try to move it, but start it where you want it to be

Well...i have tried to move it before it start as well (during cv::nameWindow declare).. still cannot.. codes are in the viewer.cc in run() function. i dont think, there are other codes for the window in other files.