UZ-SLAMLab/ORB_SLAM3 #858: Monocular mode, Map point coordinates become small after turning left

AiteYx, 2024-01-19:
Using my own video & camera parameters. Only monocular mode activated.
1、The map is simple. Go straight. - Turn left. - Go straight
2、The visualization works well the first time you go straight
3、The left turn is not bad either
4、After turning around, the map points and keyframes are very crowded. The coordinates are getting smaller

I guess the possible reason：I slows down on the left turn. After the right turn, I slowly picked up speed. But the system didn't respond

I want to know how to solve this problem. Except for uniform speed all the way (whether turning or going straight), in the case of pure monocular.

AiteYx, 2024-01-19:
Seems to have lost depth, not speed

Styazoua, 2024-01-19:
I have noticed this behaviour too. And I think it could be a change of speed. (Just an idea)
I think you could solve this by using Monocular-Inertial.

AiteYx, 2024-01-22:
> I have noticed this behaviour too. And I think it could be a change of speed. (Just an idea) I think you could solve this by using Monocular-Inertial.

Thank you.

AiteYx, 2024-01-22:
> I have noticed this behaviour too. And I think it could be a change of speed. (Just an idea) I think you could solve this by using Monocular-Inertial.

I recorded in an underground parking lot, resulting in sharp corners at the farthest and nearest points in the image. It may be difficult for the system to distinguish these corner distance information from the monocular camera.

But in this [example](https://www.bilibili.com/video/BV1XL4y1n7fQ/?spm_id_from=333.788.recommend_more_video.0&vd_source=e6aa48a4710df9cd2d42ee7386fe7bcb), monocular works very well.