UZ-SLAMLab/ORB_SLAM3 #861: How do I visualize the data after running SLAM?

crose72, 2024-01-28:
I have compiled ORB SLAM3 and all of its prerequisites, and then I downloaded and ran the Monocular EuRoC MH01 example successfully.  It generated "CameraTrajectory.txt" and "KeyFrameTrajectory.txt".  Now I would like to visualize the map but I don't know how to do that.  The instructions in this repo say to use "./euroc_eval_examples" shell script but that file doesn't exist.

How do I visualize this data?

Thanks in advance for your help!

snehabanda30, 2024-11-07:
Did any of you guys end up figuring this out?

crose72, 2024-11-07:
I did, give me some time to write up the steps I took, and share some code I wrote to help it work.  Feel free to follow up.

yusufersullu, 2025-05-28:
Do you have any updates on visualizing a live video feed?