UZ-SLAMLab/ORB_SLAM3 #863: Tracking lost when running stereo/mono mode in EuRoC dataset

ananyaverma2, 2024-02-01:
I am getting the following error "Fail to track local map" while running stereo_euroc. I tried changing the ORB parameters in the yaml file, but it doesn't seem to work.
[image]

ananyaverma2, 2024-02-07:
Tweaking the parameters solved the issue.

closed 2024-02-07 by ananyaverma2

MacintoshChen, 2024-05-08:
Hello, may I ask which parameters have you changed? I'm running into a similar issue.

ananyaverma2, 2024-05-09:
I changed a few of them but the ones that affected the tracking the most were to increase the number of features detected and decrease the threshold of ORBmatcher.

ashwinjoseph95, 2024-07-04:
Hey @ananyaverma2 , I have an issue with getting the example mentioned on the git page with the EuRoC dataset working,
I was following the [tutorial](https://github.com/UZ-SLAMLab/ORB_SLAM3?tab=readme-ov-file) and was able to build using ./build_ros.sh on Ubuntu 20.04LTS having ROS Noetic.
As per the tutorial the next step is a rosrun command like 'rosrun ORB_SLAM3 Mono PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE'
I then tried the EuRoC example command:
rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples/Stereo-Inertial/EuRoC.yaml true
but executing it gives me the error "[rosrun] Couldn't find executable named Stereo_Inertial below /home/knnas/catkin_ws/src/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3" I don't really know why it is searching in the Examples_old folder =, can you give me any leads on what am I doing wrong here?

[image]