UZ-SLAMLab/ORB_SLAM3 #866: Consistent Loss in Tracking after X Frames in Imu Mono mode

geoeo, 2024-03-06:
Hello,

I have a problem with the tracking from aerial images. It seems to be that after x frames the IMU initialization kicks in and I have a short loss in tracking. (but Im not sure for it is this!) The map may recover or may not.

Have you also observed this?
What can I do to mitigate this? I tried to increase chi2 noise values and reduce some inlier count. This has helped, but the loss still occurs.

In the images you can see a white space where the tracking was lost, but the map was able to recover.

[image]
[image]

aboluo, 2024-04-18:
you can check the timestamp consistency between the picture and the imu data

geoeo, 2024-10-31:
I found a lot of timestamp issue in the original repo. you can try this one, it works for me now
https://github.com/geoeo/ORB_SLAM3_V1.0

geoeo, 2024-11-02:
I think there were hardcoded offsets in the code. Check Tracking.cc for starters. But I also upgraded to a 400Hz Imu.