UZ-SLAMLab/ORB_SLAM3 #882: i had a problem when loading atlas maps:terminate called after throwing an instance of 'boost::archive::archive_exception'   what():  input stream error Aborted (core dumped)

aboluo, 2024-04-18:
for example, when i load the map in osa format file that was saved before, it reports:terminate called after throwing an instance of 'boost::archive::archive_exception'
  what():  input stream error
Aborted (core dumped)
has anyone encountered a similar problem and how did you solve it,  many thanks

RaduCorcodel, 2024-07-30:
@aboluo I have exactly the same issue when trying to load a previously saved map. Have you found a fix for this?

RaduCorcodel, 2024-07-30:
@arjunskumar size on disk is 92MB

RaduCorcodel, 2024-08-01:
Alas, no. I'm not using ROS. Should it be in the `.ros` folder? I did a little poking around and it looks like it exits in the middle of writing the `.osa` file. I start to believe the archive is corrupted

tianyilim, 2025-09-26:
Hi, were you able to resolve this?

tianyilim, 2025-09-27:
Aside from the solutions in [this comment](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/443#issuecomment-1154814254) in #443, I found a potential other source of the bug.

In [`src/System.cc`](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/System.cc#L1411-L1413):

```cpp
        string pathSaveFileName = "./";
        pathSaveFileName = pathSaveFileName.append(mStrSaveAtlasToFile);
        pathSaveFileName = pathSaveFileName.append(".osa");
```

If you specified an *absolute* path in your config file (`System.SaveAtlasToFile`), this code will nonetheless attempt to create the atlas file as a relative path. This then leads to [the later line](https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/System.cc#L1436) creating a Boost Archive to fail, because (possibly multiple folders) have to be created before the Atlas `.osa` file is saved.

If you change the code block above to:
```cpp
        string pathSaveFileName = "";
        pathSaveFileName = pathSaveFileName.append(mStrSaveAtlasToFile);
        pathSaveFileName = pathSaveFileName.append(".osa");
```