UZ-SLAMLab/ORB_SLAM3 #883: How to set initial

ProKaroly, 2024-04-18:


ProKaroly, 2024-04-18:
Does anyone have experience with setting the pose of the initial keyframe?

I'm attempting to translate the entire map using a translation vector. I've incorporated this translation during the MonocularMapInitialization process and have also translated the initial MapPoints. However, after the translation, the Tracking process fails. I suspect that the TrackReferenceKeyFrame() function, which estimates the pose of the third frame, might be encountering inaccuracies in the estimated pose, leading to the "Fail to track" error.

basedball, 2024-04-24:
If you only want to translate the output pose returned by TrackMonocular(), you could just replace the return statement with your code: 
```
Eigen::Vector3f translationVector(0.0, 0.0, 210.0); // Adjust the values as needed
Sophus::SE3f currentFramePose = Tcw * Sophus::SE3f(Eigen::Quaternionf::Identity(), translationVector);
return currentFramePose;
```

If you actually want to shift everything in the map, you could try passing a translation vector into the Frame constructor used in TrackMonocular and set the pose at the top of the constructor. Changing the end of MonocularInitialization() to:
```
// Set the pose of the current frame
mCurrentFrame.SetPose(mInitialFrame.GetPose() * Tcw);
CreateInitialMapMonocular();
```
should set the second Frame correctly, and you won't have to manually change the MapPoints. Hope this helps.

ProKaroly, 2024-04-29:
I would like to connect the maps. So you second approach would be better, but unfortunately during initialization the Triangulation() function do not take care of the pose of the initial Frame. It estimate the relative Transformation and use it as the pose of the second Frame.

Narek-939, 2024-05-07:
> If you only want to translate the output pose returned by TrackMonocular(), you could just replace the return statement with your code:
>
> ```
> Eigen::Vector3f translationVector(0.0, 0.0, 210.0); // [...]

Hi thank you for your reply, I have tried second version to merge maps, but got the same when wrote as bellow

mInitialFrame.SetPose(Sophus::SE3f()); 
mCurrentFrame.SetPose(Tcw);
CreateInitialMapMonocular();

please can you explain me this part, how to merge maps correctly, if I get recently relocalization