UZ-SLAMLab/ORB_SLAM3 #886: points and path disappear and start over, every minute in ORB SLAM3 Ros

Spnd77, 2024-04-24:
Hi everyone.
I Start working on ORBSLAM3, I used of:
Ubuntu 20.04
Camera: RealSense T265
Jetson xavier jetPack 5.1.2
L4T: 35.4.1
RealSense SDK 2.50.0
Open CV: 4.5.4
when I run ORBSLAM3 it can detect points and path in Orbslam Map viewer, But when I move camera, every minute points and path disappear and it start again... how I can fix this issue? it's need special config to can show path and points realtime??

Problem 2:
I read calibration pdf, it say you should configure  TBC1 and TC1C2 in stereo Inertial, and it say Camera2(c2) is Right Camera in T265
but in RealSense_T265.yaml is C2 as left Camera:

# Transformation matrix from right camera to left camera
Stereo.T_c1_c2: !!opencv-matrix
  rows: 4
  cols: 4
  dt: f
  data: [ 0.999983,   0.00445005, 0.00385861,  0.0636739954352379,
         -0.00443664, 0.999984,  -0.00347621, -0.000252007856033742,
         -0.00387402, 0.00345903, 0.999986,   -8.87895439518616e-05,
          0.0, 0.0, 0.0, 1.0]
          
which is Correct? please Help me for these issues.

basedball, 2024-04-24:
What are the messages being output to the console when it resets?

Spnd77, 2024-04-27:
Thank you @basedball 
I have these lines in terminal:
Input sensor was set to: Stereo-Inertial
Loading settings from /home/vn/catkin_ws/src/orb_slam3_ros/config/Stereo-Inertial/Realsense_T265.yaml
	-Loaded camera 1
	-Loaded camera 2
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
	-Loaded IMU calibration
	-Loaded ORB settings
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Kannala-Brandt): [ 283.795 284.952 424.865 396.591 -0.0027128 0.0366888 -0.0354853 0.0059001 ]
	-Camera 2 parameters (Kannala-Brandt): [ 284.718 285.687 430.758 396.512 -0.00173706 0.0350818 -0.0336628 0.00533116 ]
	-Original image size: [ 848 , 800 ]
	-Current image size: [ 848 , 800 ]
	-Sequence FPS: 30
	-Stereo baseline: 0.0636746
	-Stereo depth threshold : 40
	-Camera 1 overlapping area: [ 0 , 848 ]
	-Camera 2 overlapping area: [ 0 , 848 ]
	-Gyro noise: 0.001
	-Accelerometer noise: 0.01
	-Gyro walk: 1e-07
	-Accelerometer walk: 1e-07
	-IMU frequency: 200
	-Features per image: 1200
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
[ INFO] [1714192847.084943190]: rviz version 1.14.20
[ INFO] [1714192847.085183415]: compiled against Qt version 5.12.8
[ INFO] [1714192847.085296518]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1714192847.117307769]: Forcing OpenGl version 0.
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 2 cameras in the atlas
Camera 0 is fisheye
Camera 2 is fisheye
[ INFO] [1714192848.612980547]: Stereo is NOT SUPPORTED
[ INFO] [1714192848.613352949]: OpenGL device: NVIDIA Tegra Xavier (nvgpu)/integrated
[ INFO] [1714192848.613483078]: OpenGl version: 4.6 (GLSL 4.6).
Starting the Viewer
not IMU meas
Map point vector is empty!
not IMU meas
Map point vector is empty!
not enough acceleration
Map point vector is empty!
First KF:0; Map init KF:0
New Map created with 177 points
[ INFO] [1714192857.483827252]: Finished waiting for tf, waited 11.007096 seconds
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 3
mnInitialFrameId = 0
14 Frames set to lost
not IMU meas
Map point vector is empty!
First KF:12; Map init KF:0
New Map created with 147 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 18
mnInitialFrameId = 16
34 Frames set to lost
not IMU meas
Map point vector is empty!
First KF:22; Map init KF:12
New Map created with 260 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 39
mnInitialFrameId = 37
55 Frames set to lost
not IMU meas
Map point vector is empty!
First KF:34; Map init KF:22
New Map created with 219 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 61
mnInitialFrameId = 59
77 Frames set to lost
not IMU meas
Map point vector is empty!
First KF:45; Map init KF:34
New Map created with 71 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 84
mnInitialFrameId = 82
80 Frames set to lost
not IMU meas
Map point vector is empty!
First KF:46; Map init KF:45
New Map created with 66 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 88
mnInitialFrameId = 86
82 Frames set to lost
not IMU meas
Map point vector is empty!
First KF:47; Map init KF:46
New Map created with 111 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 91
mnInitialFrameId = 89
85 Frames set to lost
not IMU meas
Map point vector is empty!
First KF:48; Map init KF:47
New Map created with 178 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 95
mnInitialFrameId = 93
102 Frames set to lost
not IMU meas
Map point vector is empty!

SrOdranoel, 2025-03-19:
any solution?