UZ-SLAMLab/ORB_SLAM3 #892: Dose Stereo-image can reconstruct dense point cloud?

hplegend, 2024-05-12:
For help. I just want to known that there is any solution using stereo-image construct dense map?

WFram, 2024-06-11:
ORB-SLAM doesn't construct dense maps; check densification approaches, like OpenMVS. (one camera is enough). ORB-SLAM3 can provide good enough poses to run OpenMVS (especially in Stereo-Inertial mode).

hplegend, 2024-07-09:
> ORB-SLAM doesn't construct dense maps; check densification approaches, like OpenMVS. (one camera is enough). ORB-SLAM3 can provide good enough poses to run OpenMVS (especially in Stereo-Inertial mode).

Can you provide some examples？ I try to export ORB-SLAM camera pose and mapPoints as input of OpenMVS，but it failed to obtain dense point cloud.