UZ-SLAMLab/ORB_SLAM3 #898: Segmentation Fault issue during runtime -  Eigen::internal::cast_impl

mjunsen123, 2024-05-22:
Running ORB-SLAM3 with ROS2 Humble wrapper in Ubuntu 22.04. The program run but sometimes it crash with segmentation fault. See below debug backtrace:
`Thread 29 "stereo-inertial" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff3d9ff000 (LWP 274445)]
0x00007ffff7cae09b in Eigen::internal::cast_impl<float, double, void>::run (x=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/functors/AssignmentFunctors.h:24
24        EIGEN_DEVICE_FUNC EIGEN_STRONG_INLINE void assignCoeff(DstScalar& a, const SrcScalar& b) const { a = b; }
(gdb) backtrace
#0  0x00007ffff7cae09b in Eigen::internal::cast_impl<float, double, void>::run(float const&) (x=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/functors/AssignmentFunctors.h:24
#1  Eigen::internal::cast<float, double>(float const&) (x=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/MathFunctions.h:462
#2  Eigen::internal::scalar_cast_op<float, double>::operator()(float const&) const (a=<optimized out>, this=<optimized out>)
    at /usr/include/eigen3/Eigen/src/Core/functors/UnaryFunctors.h:163
#3  Eigen::internal::unary_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const>, Eigen::internal::IndexBased, double>::coeff(long) const (index=<optimized out>, this=<optimized out>) at /usr/include/eigen3/Eigen/src/Core/CoreEvaluators.h:589
#4  Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3> >, Eigen::internal::evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> >, Eigen::internal::assign_op<double, double>, 0>::assignCoeff(long)
    (index=0, this=<synthetic pointer>) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:660
#5  Eigen::internal::copy_using_evaluator_LinearTraversal_CompleteUnrolling<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3> >, Eigen::internal::evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> >, Eigen::internal::assign_op<double, double>, 0>, 0, 9>::run(Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3> >, Eigen::internal::evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> >, Eigen::internal::assign_op<double, double>, 0>&) (kernel=<synthetic pointer>...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:247
#6  Eigen::internal::dense_assignment_loop<Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3> >, Eigen::internal::evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> >, Eigen::internal::assign_op<double, double>, 0>, 1, 2>::run(Eigen::internal::generic_dense_assignment_kernel<Eigen::internal::evaluator<Eigen::Matrix<double, 3, 3, 0, 3, 3> >, Eigen::internal::evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> >, Eigen::internal::assign_op<double, double>, 0>&)
    (kernel=<synthetic pointer>...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:528
#7  Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, 3, 3, 0, 3, 3>&, Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> const&, Eigen::internal::assign_op<double, double> const&) (dst=..., src=..., func=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:785
#8  0x00007ffff7ca5edb in Eigen::internal::Assignment<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const>, Eigen::internal::assign_op<double, double>, Eigen::internal::Dense2Dense, void>::run(Eigen::Matrix<double, 3, 3, 0, 3, 3>&, Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> const&, Eigen::internal::assign_op<double, double> const&)
    (func=..., src=..., dst=...) at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:948
#9  Eigen::internal::call_assignment_no_alias<Eigen::Matrix<double, 3, 3, 0, 3, 3>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, 3, 3, 0, 3, 3>&, Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> const&, Eigen::internal::assign_op<double, double> const&) (func=..., src=..., dst=...)
    at /usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:890
#10 Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::_set_noalias<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> >(Eigen::DenseBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> > const&)
    (other=..., this=0x7ffe96bd8288) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:797
#11 Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 3, 0, 3, 3> >::_init1<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> >(Eigen::DenseBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> > const&) (other=..., this=0x7ffe96bd8288)
    at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:883
#12 Eigen::Matrix<double, 3, 3, 0, 3, 3>::Matrix<Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> >(Eigen::CwiseUnaryOp<Eigen::internal::scalar_cast_op<float, double>, Eigen::Matrix<float, 3, 3, 0, 3, 3> const> const&) (x=..., this=0x7ffe96bd8288)
    at /usr/include/eigen3/Eigen/src/Core/Matrix.h:332
#13 ORB_SLAM3::EdgeInertial::EdgeInertial(ORB_SLAM3::IMU::Preintegrated*) (this=0x7ffe96bd7ea0, pInt=0x0) at /home/jyee/ORB_SLAM3/src/G2oTypes.cc:494
#14 0x00007ffff7c071e0 in ORB_SLAM3::Optimizer::PoseInertialOptimizationLastFrame(ORB_SLAM3::Frame*, bool)
    (pFrame=pFrame@entry=0x7fff5d64f020, bRecInit=bRecInit@entry=false) at /home/jyee/ORB_SLAM3/src/Optimizer.cc:5059
#15 0x00007ffff7b7c2e4 in ORB_SLAM3::Tracking::TrackLocalMap() (this=0x7fff5d64f010) at /home/jyee/ORB_SLAM3/src/Tracking.cc:3003
#16 0x00007ffff7b83408 in ORB_SLAM3::Tracking::Track() (this=0x7fff5d64f010) at /home/jyee/ORB_SLAM3/src/Tracking.cc:2145
--Type <RET> for more, q to quit, c to continue without paging--c
#17 0x00007ffff7b83dcc in ORB_SLAM3::Tracking::GrabImageStereo(cv::Mat const&, cv::Mat const&, double const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (this=this@entry=0x7fff5d64f010, imRectLeft=..., imRectRight=..., timestamp=@0x7fff3d9ec6c8: 1716372586.3878787, filename="") at /home/jyee/ORB_SLAM3/src/Tracking.cc:1513
#18 0x00007ffff7b2c728 in ORB_SLAM3::System::TrackStereo(cv::Mat const&, cv::Mat const&, double const&, std::vector<ORB_SLAM3::IMU::Point, std::allocator<ORB_SLAM3::IMU::Point> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (this=0x7fffffffc920, imLeft=..., imRight=..., timestamp=@0x7fff3d9ec6c8: 1716372586.3878787, vImuMeas=std::vector of length 0, capacity 0, filename="") at /home/jyee/ORB_SLAM3/src/System.cc:448
#19 0x00005555555702b4 in StereoInertialNode::SyncWithImu() (this=0x55555566a490) at /home/jyee/ros2_ws/src/ORB_SLAM3_ROS2/src/stereo-inertial/stereo-inertial-node.cpp:210
#20 0x00007ffff4894253 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#21 0x00007ffff4603ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#22 0x00007ffff4695850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81`

nesquik011, 2024-07-28:
did you fix it ?

mjunsen123, 2024-07-29:
Yes, I do found that the issue is on camera end. If you are using ROS2 with your own camera, check whether there is frame drop from camera / IMU.