UZ-SLAMLab/ORB_SLAM3 #902: The bug in the code within the Optimizer::FullInertialBA function

zhangchao-nreal, 2024-05-31:
https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/Optimizer.cc#L684C19-L684C83
The rightIndex is the actual index plus mvKeysRight.size(), so the condition rightIndex < pKFi->mvKeysRight.size() can never be satisfied. This will result in the right camera constraints not being added.

jeremysalwen, 2024-07-27:
Why do you say that the right index is the actual index plus mvKeysRight.size()?

zhangchao-nreal, 2024-12-11:
rightIndex -= pKFi->NLeft;
Eigen::Matrix<double,2,1> obs;
kpUn = pKFi->mvKeysRight[rightIndex];

I made a mistake in understanding this. From the code above, it can be seen that rightIndex is actually the index plus mvKeys.size(), which is Nleft. Therefore, the condition rightIndex < pKFi->mvKeysRight.size() is incorrect. This will result in the right camera constraints not being added.