UZ-SLAMLab/ORB_SLAM3 #903: make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:554: ../lib/libORB_SLAM3.so] Error 1 make[1]: *** [CMakeFiles/Makefile2:711: CMakeFiles/ORB_SLAM3.dir/all] Error 2

AmazingDuDuDu, 2024-06-05:
I'm trying to run `./build.sh` to build the Thirdparty libraries and ORB-SLAM3 and get this error.

I have run the commands in build.sh one by one. The error occurs when I run `make -j4` to build ORB-SLAM3.

Does anyone know how to solve it? Thanks in advance.

Kanixre, 2024-07-07:
I am having the same issue. Were you able to solve it? Thank you.

ekrzina, 2024-07-26:
Did you download all necessary dependencies? That would be Eigen, Pangolin, g2o and DBoW2 (git clone). These also have some dependencies, like libboost-serialization-dev and libssl-dev (apt install).

Also, for CMakeLists.txt, make sure to type the following before building ORB_SLAM3:
```
sed -i 's/++11/++14/g' CMakeLists.txt
chmod +x build.sh
./build.sh
```

This will replace C++11 with C++14 as standard version.

MarcoRovell, 2024-07-26:
if we download only the most recent versions of all these dependencies, could some things be deprecated? I get tons of warnings after doing this?

ekrzina, 2024-07-27:
I've downloaded the newest versions of all dependencies very recently (a week ago), and only had trouble with the c++ version, which can be changed with the sed command I've commented above. So I don't think this should be the case.

Also, when building again, make sure to delete the old build directory and create a new one as it could read cache. If nothing works, replace the c++ to version 14 in all other CMakeLists and build those anew. All need to be built correctly before ORB_SLAM3 cab be built.

Hope this helps.

lavishuse1, 2025-02-06:
can you resolve this error
/home/lavish/ORB_SLAM3/Thirdparty/Sophus/test/core/test_sim2.cpp:214:24:
/usr/lib/gcc/x86_64-linux-gnu/13/include/xmmintrin.h:941:23: error: array subscript ‘__m128_u[0]’ is partly outside array bounds of ‘Sophus::Vector<float, 2, 0> [1]’ {aka ‘Eigen::Matrix<float, 2, 1> [1]’} [-Werror=array-bounds=]
  941 |   return *(__m128_u *)__P;

ZeinBarhoum, 2025-03-09:
@lavishuse1 
I solved this by adding `add_definitions(-DEIGEN_DONT_ALIGN)` in `Thirdparty/Sophus/CMakeLists.txt b/Thirdparty/Sophus/CMakeLists.txt`
```
 # Find Eigen 3 (dependency)
add_definitions(-DEIGEN_DONT_ALIGN) #here 
 find_package(Eigen3 3.3.0 REQUIRED)
```