UZ-SLAMLab/ORB_SLAM3 #904: Linking error in building

SPGC, 2024-06-05:
```
[ 47%] Linking CXX shared library ../lib/libORB_SLAM3.so
/usr/bin/ld: cannot find -lepoxy::epoxy: No such file or directory
/usr/bin/ld: cannot find -lOpenGL::EGL: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:600: ../lib/libORB_SLAM3.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:148: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```
All dependencies listed in ReadMe are installed. Target C++ version changed to 14, otherwise there were errors connected to Pangoline

SPGC, 2024-06-06:
Solution found. To solve this problem it's need to download [file](https://github.com/stevenlovegrove/Pangolin/blob/master/cmake/Findepoxy.cmake) from Pangolin repository and put it in project dir, after that in CMakeList.txt add:
```
set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake_modules/)
find_package(epoxy REQUIRED)
include_directories(
${epoxy_INCLUDE_DIRS}
)
target_link_libraries(${PROJECT_NAME}
${epoxy_LIBRARIES}
)
```

closed 2024-06-06 by SPGC

ouguangjun, 2024-06-06:
thanks, it works

xwy-bit, 2024-06-06:
> Solution found. To solve this problem it's need to download [file](https://github.com/stevenlovegrove/Pangolin/blob/master/cmake/Findepoxy.cmake) from Pangolin repository and put it in project dir, after that in CMakeList.txt add:
> 
> ```
> set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake_modules/)
> 
> ...
> 
> find_package(OpenGL)
> find_package(epoxy REQUIRED)
> 
> include_directories(
> ${PROJECT_SOURCE_DIR}
> ${PROJECT_SOURCE_DIR}/include
> ${PROJECT_SOURCE_DIR}/include/CameraModels
> ${PROJECT_SOURCE_DIR}/Thirdparty/Sophus
> ${EIGEN3_INCLUDE_DIR}
> ${Pangolin_INCLUDE_DIRS}
> ${epoxy_INCLUDE_DIRS}
> )
> 
> ...
> 
> target_link_libraries(${PROJECT_NAME}
> ${OpenCV_LIBS}
> ${EIGEN3_LIBS}
> ${Pangolin_LIBRARIES}
> ${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so
> ${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib/libg2o.so
> -lboost_serialization
> -lcrypto
> ${epoxy_LIBRARIES}
> )
> ```

Well done, it works. By the way, the [file](https://github.com/stevenlovegrove/Pangolin/blob/master/cmake/Findepoxy.cmake) should be added to the `ORB_SLAM3/cmake_modules` dir.

SPGC, 2024-06-06:
> Well done, it works. By the way, the [file](https://github.com/stevenlovegrove/Pangolin/blob/master/cmake/Findepoxy.cmake) should be added to the `ORB_SLAM3/cmake_modules` dir.

Yes, sure, forgot to mention. Fixed this, thank you!