UZ-SLAMLab/ORB_SLAM3 #913: Unable to build ORB_SLAM3 with ROS using ./build_ros.sh

rioter1, 2024-07-01:
I have installed ros-noetic-desktop-full
Paths for ros_root and ros_package are
echo $ROS_ROOT
/opt/ros/noetic/share/ros
echo $ROS_PACKAGE_PATH
/opt/ros/noetic/share:~/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3

after running ./build_ros.sh, I get 

[rosbuild] Building package ORB_SLAM3
Failed to invoke /opt/ros/noetic/bin/rospack deps-manifests ORB_SLAM3
ModuleNotFoundError: No module named 'rosdep2'
[rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?

CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:129 (message):
  

  Failed to invoke rospack to get compile flags for package 'ORB_SLAM3'.
  Look above for errors from rospack itself.  Aborting.  Please fix the
  broken dependency!

Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
  CMakeLists.txt:5 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/antpc/orbslam/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

I can solve the rosdep error by doing source /opt/ros/noetic/setup.bash
but this changes my ROS_PACKAGE_PATH to 
echo $ROS_PACKAGE_PATH 
/opt/ros/noetic/share
and gives new error 
[rosbuild] Error from directory check: /opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/antpc/orbslam/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3
1
Traceback (most recent call last):
  File "/opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:99 (message):
  [rosbuild] rospack found package "ORB_SLAM3" at "", but the current
  directory is "~/orbslam/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3".
  You should double-check your ROS_PACKAGE_PATH to ensure that packages are
  found in the correct precedence order.
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:5 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/antpc/orbslam/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

If I do export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/orbslam/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3 as is the error above, I again get the rosdep2 error Error: could not find python module 'rosdep2.rospack'

Its like I am going in circles.

rioter1, 2024-07-01:
echo $ROS_PACKAGE_PATH
/home/antpc/orbslam/ORB_SLAM3/Examples_old/ROS
echo $ROS_ROOT
/opt/ros/noetic/share/ros

[rosbuild] Building package ORB_SLAM3
Failed to invoke /opt/ros/noetic/bin/rospack deps-manifests ORB_SLAM3
[rospack] Error: package 'ORB_SLAM3' depends on non-existent package 'roscpp' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'ORB_SLAM3' depends on non-existent package 'tf' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'ORB_SLAM3' depends on non-existent package 'sensor_msgs' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'ORB_SLAM3' depends on non-existent package 'image_transport' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'
[rospack] Error: package 'ORB_SLAM3' depends on non-existent package 'cv_bridge' and rosdep claims that it is not a system dependency. Check the ROS_PACKAGE_PATH or try calling 'rosdep update'

CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:129 (message):
  

  Failed to invoke rospack to get compile flags for package 'ORB_SLAM3'.
  Look above for errors from rospack itself.  Aborting.  Please fix the
  broken dependency!

Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
  CMakeLists.txt:5 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/antpc/orbslam/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

All packages are installed and are the latest version

khshmt, 2024-07-01:
Did you run "rosdep update" after finishing the ros installation

rioter1, 2024-07-01:
> Did you run "rosdep update" after finishing the ros installation

yes

rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Skip end-of-life distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Add distro "iron"
Skip end-of-life distro "jade"
Add distro "jazzy"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Skip end-of-life distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /home/antpc/.ros/rosdep/sources.cache

khshmt, 2024-07-01:
I am sorry but just to make sure, you ran "rosdep update" after "sudo rosdep init" after installation.

rioter1, 2024-07-01:
> I am sorry but just to make sure, you ran "rosdep update" after "sudo rosdep init" after installation.

yes I ran rosdep update after I ran sudo rosdep init

khshmt, 2024-07-01:
I recommend that you clone any cpp ros package and try to build it in a catkin workspace to ensure that you have a clean ros installation.

rioter1, 2024-07-01:
> I recommend that you clone any cpp ros package and try to build it in a catkin workspace to ensure that you have a clean ros installation.

will give that a shot. Thanks for your responses Khaled!

rioter1, 2024-07-02:
I resolved the errors by purging ros-noetic from my system and reinstalling it. didn't require catkin_make just used the standard ros install as per http://wiki.ros.org/noetic/Installation/Ubuntu. Thanks Khaled!