UZ-SLAMLab/ORB_SLAM3 #915: Stereo kitti Yaml

nesquik011, 2024-07-09:
i have a problem , i am running stereo kitti example , and i used this yaml 

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 718.856
Camera.fy: 718.856
Camera.cx: 607.1928
Camera.cy: 185.2157

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.bFishEye: 0

Camera.width: 1241
Camera.height: 376

# Camera frames per second 
Camera.fps: 10.0

# stereo baseline times fx
Camera.bf: 386.1448

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 35.0

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 2000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.6
Viewer.KeyFrameLineWidth: 2
Viewer.GraphLineWidth: 1
Viewer.PointSize: 2
Viewer.CameraSize: 0.7
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -100
Viewer.ViewpointZ: -0.1
Viewer.ViewpointF: 2000

C:\Users\dream>"E:\orb slam 3 test\orbslam test 4\x64\Release\orbslam test 1.exe"  E:/vcpkg/ORB_SLAM3/Vocabulary/ORBvoc.txt E:/vcpkg/ORB_SLAM3/Examples/Stereo/KITTI00-02.yaml D:/dataset/sequences/00

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. G├│mez, Jos├⌐ M.M. Montiel and Juan D. Tard├│s, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Ra├║l Mur-Artal, Jos├⌐ M.M. Montiel and Juan D. Tard├│s, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo
Loading settings from E:/vcpkg/ORB_SLAM3/Examples/Stereo/KITTI00-02.yaml
        -Loaded camera 1
        -Loaded camera 2
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
        -Loaded image info
        -Loaded ORB settings
Viewer.ViewpointY parameter must be a real number, aborting...

NOTE : Windows built with zero errors

nesquik011, 2024-07-11:
i used opencv 4.8.1 , and 4.5.3 still the same problem , 3 different yamls 

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

# ===> NOTICE: Add camera model here <===
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 718.856
Camera.fy: 718.856
Camera.cx: 607.1928
Camera.cy: 185.2157

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.bFishEye: 0

Camera.width: 1241
Camera.height: 376

# Camera frames per second 
Camera.fps: 10.0

# stereo baseline times fx
Camera.bf: 386.1448

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 40.0

# ============================================================

# ===> NOTICE: Add camera parameters here <===
LEFT.height: 1241
LEFT.width: 376
LEFT.D: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data: [0.0, 0.0, 0.0, 0.0, 0.0]
LEFT.K: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [718.856, 0.0, 607.1928, 0.0, 718.856, 185.2157, 0.0, 0.0, 1.0]
LEFT.R:  !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
LEFT.P:  !!opencv-matrix
   rows: 3
   cols: 4
   dt: d
   data: [7.188560000000e+02, 0.000000000000e+00, 6.071928000000e+02, 0.000000000000e+00, 0.000000000000e+00, 7.188560000000e+02, 1.852157000000e+02, 0.000000000000e+00, 0.000000000000e+00, 0.000000000000e+00, 1.000000000000e+00, 0.000000000000e+00]

RIGHT.height: 1241
RIGHT.width: 376
RIGHT.D: !!opencv-matrix
   rows: 1
   cols: 5
[... 162 lines cut ...]
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 12
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.6
Viewer.KeyFrameLineWidth: 2
Viewer.GraphLineWidth: 1
Viewer.PointSize:2
Viewer.CameraSize: 0.7
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -100
Viewer.ViewpointZ: -0.1
Viewer.ViewpointF: 2000

nesquik011, 2024-07-11:
@richard-elvira @jdtardos @jj-gomez @ccamposm  kindly try to help me

closed 2024-07-17 by nesquik011

nesquik011, 2024-07-22:
@kimzheng1204 yes i passed this problem , dont be upset i dont remember what i ready did , right now i am in a way bigger problem , do you have discord ? we can chat there so i can fix yours

nesquik011, 2024-07-22:
i apologize guys for not posting how i fixed this problem 👎

casimirfeldmann, 2024-11-14:
The fix is easy: 
"Viewer.ViewpointY: -100" needs a decimal. Change it to "Viewer.ViewpointY: -100.0"

nesquik011, 2024-11-14:
mostly i guess you are right , basically it wont be a true number unless you do the .0 in whatever number  , so for anyone looking for this fix  you must  do .0 or . whatever number

SrOdranoel, 2025-02-24:
Hello I am having this problem bellow i belive is something with the yaml file can we talk about:
```
First KF:33; Map init KF:32
New Map created with 17 points
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 173
mnInitialFrameId = 157
36 Frames set to lost
not IMU meas
First KF:34; Map init KF:33
New Map created with 23 points
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 176
mnInitialFrameId = 174
37 Frames set to lost
not IMU meas

```

nesquik011, 2025-03-09:
are you using stereo ?

SrOdranoel, 2025-03-10:
Hi @nesquik011   Stereo -Inertial and i am using two cameras FLIR point gray 1600x1200 and a stim300. You have any tip or idea how i can solve?