UZ-SLAMLab/ORB_SLAM3 #924: ./build_ros.sh error,在cmake阶段出现问题，疑似python的问题，目前毫无头绪

muqianjx, 2024-08-22:
```
muqian@muqian-virtual-machine:~/catkin_ws/src/ORB_SLAM3$ ./build_ros.sh 
Building ROS nodes
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

-- Found PythonInterp: /usr/bin/python (found version "2.7.17") 
[rosbuild] Building package ORB_SLAM3
Failed to invoke /opt/ros/melodic/bin/rospack deps-manifests ORB_SLAM3
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospack.py", line 60, in init_rospack_interface
    lookup = _get_default_RosdepLookup(Options())
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 611, in create_default
    matcher = DataSourceMatcher.create_default(os_override=os_override)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 290, in create_default
    os_name, os_version, os_codename = os_detect.detect_os()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 608, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s" % attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['zorin', 'windows', 'nixos', 'clearlinux', 'ubuntu', 'slackware', 'rocky', 'rhel', 'raspbian', 'qnx', 'pop', 'osx', 'sailfishos', 'tizen', 'conda', 'oracle', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'opensuse', 'openembedded', 'neon', 'mx', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora-asahi', 'fedora', 'elementary', 'elementary', 'debian', 'cygwin', 'euleros', 'centos', 'manjaro', 'buildroot', 'arch', 'amazon', 'alpine', 'almalinux']
[rospack] Error: could not call python function 'rosdep2.rospack.init_rospack_interface'

CMake Error at /opt/ros/melodic/share/ros/core/rosbuild/public.cmake:129 (message):
  

  Failed to invoke rospack to get compile flags for package 'ORB_SLAM3'.
  Look above for errors from rospack itself.  Aborting.  Please fix the
  broken dependency!

Call Stack (most recent call first):
  /opt/ros/melodic/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/muqian/catkin_ws/src/ORB_SLAM3/Examples/ROS/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** 没有指明目标并且找不到 makefile。 停止。
```

arjunskumar, 2024-08-22:
@muqianjx 
try this
`sudo apt-get install python3-rospkg python3-rosdep2`

muqianjx, 2024-08-27:
> @muqianjx try this `sudo apt-get install python3-rospkg python3-rosdep2`
Thanks for your response, but I have fixed the issue with another method.
The problem is：raise OsNotDetected("Could not detect OS, tried %s" % attempted)
So I fixed it and marked it as closed.

muqianjx, 2024-08-27:
### Problem:`raise OsNotDetected("Could not detect OS, tried %s" % attempted)`
### Method
Checking your `/etc/lsb-release`,Observe if it becomes what it looks like below
```
DISTRIB_ID=Kylin
DISTRIB_RELEASE=V10
DISTRIB_CODENAME=kylin
DISTRIB_DESCRIPTION="Kylin V10 SP1"
DISTRIB_KYLIN_RELEASE=V10
DISTRIB_VERSION_TYPE=enterprise
DISTRIB_VERSION_MODE=normal
```
If yes,Clear the existing content and change it to the following
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.6 LTS"
```
If not,you need to look for other methods.

closed 2024-08-27 by muqianjx