UZ-SLAMLab/ORB_SLAM3 #925: T_b_c1 from left cam to imu or the other way around?

genkv, 2024-08-23:
Sorry if this seems stupid, but when you use stereo-inertial mode, in the configuration yaml, is `IMU.T_b_c1 `a transformation matrix from the left camera to imu, or from imu to the left camera?

The comments says `IMU.T_b_c1: !!opencv-matrix` but the name clearly indicates that it should be the other way around. Just like `Stereo.T_c1_c2` is a transformation from the left camera to the right camera, if `c1` stands for the left camera.

I had success with the stereo mode, but every time I use the same dataset I created (with addition to imu) on stereo-inertial, it would constantly fail. It must've been something wrong with my IMU setting

radhasaraf, 2024-08-23:
`IMU.T_b_c1` is the transformation matrix from c1 to imu
`Stereo.T_c1_c2` is the transformation matrix from c2 to c1.
(If you have a point in c2's coordinate frame(p2), then `T_c1_c2 * p2` should give you the point in c1's coordinate frame)

genkv, 2024-08-23:
> `IMU.T_b_c1` is the transformation matrix from c1 to imo
> `Stereo.T_c1_c2` is the transformation matrix from c2 to c1
> (If you have a point in c2's coordinate frame(p2), then `T_c1_c2 * p2` should give you the point in c1's coordinate frame)

Thank you! It was a bit confusing

closed 2024-08-23 by genkv