UZ-SLAMLab/ORB_SLAM3 #929: Fail to track local map!

dapao-0808, 2024-09-04:
Hi! When I use my own device in IMU+Stereo mode, I always encounter the message 'Fail to track local map!' after running for a period of time! Then reset the map, but this phenomenon does not occur in stereo mode. What are the reasons for this phenomenon?
[image]

dapao-0808, 2024-10-29:
My reason is due to excessive calibration errors between the camera and IMU

dapao-0808, 2024-10-30:
[Kalibr](https://github.com/ethz-asl/kalibr) tool can be used for calibration to ensure that the reprojection error is within 0.5 pixels, preferably between 0.1 and 0.2.

dapao-0808, 2024-11-04:
There are many factors that affect multi-sensor performance: 1. Whether the clock sources of the sensors are consistent; 2. The frequency of IMU should be at least twice as high as that of the camera, of course, the higher the better; 3. Some scenarios may have a significant impact on IMU, and it may be necessary to continue increasing the frequency of IMU or expanding its random walks and noise, etc. If you don't want to consider so much, you can try using a camera like D435i, which will save a lot of trouble

yanyongxian, 2025-03-05:
Maybe I got a solution.In my experience,when change the verbose level to [VERBOSITY_DEBUG], everything works fine.  It is pretty wierd. https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c4ab75b1cde34e5505a36ec3f9edcdc4c4/src/System.cc#L240 :blush:

SrOdranoel, 2025-03-17:
For me didnt work i sill geting the same error of the IMU you have other solution or idea what can be the problem i have calibrated my system with the kalibr and the Stereo mode works well the problem is with the IMU. Any ideas @yanyongxian @dapao-0808 @boardd ??

The problem:
```
done
mnFirstFrameId = 2345
mnInitialFrameId = 2343
2133 Frames set to lost
   End reseting! 
Preintegration is finished!! 
not IMU meas
Preintegration is finished!! 
First KF:381; Map init KF:380
New Map created with 68 points
Preintegration is finished!! 
Fail to track local map!
Track lost for less than one second...
IMU is not or recently initialized. Reseting active map...
Active map Reseting
Reseting Local Mapper...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
done
Reseting Loop Closing...
done
Reseting Database
done
mnFirstFrameId = 2348
mnInitialFrameId = 2346
2135 Frames set to lost
   End reseting! 
Preintegration is finished!! 
not IMU meas
Preintegration is finished!! 
First KF:382; Map init KF:381
New Map created with 58 points
Preintegration is finished!! 
Fail to track local map!
Track lost for less than one second...
IMU is not or recently initialized. Reseting active map...
Active map Reseting
Reseting Local Mapper...
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
done
Reseting Loop Closing...
done
Reseting Database
done
mnFirstFrameId = 2351
mnInitialFrameId = 2349
2137 Frames set to lost
   End reseting! 
Preintegration is finished!! 
```