UZ-SLAMLab/ORB_SLAM3 #941: Couldn't find executable named Mono below

wdjiang, 2024-11-07:
I run SLAM3 using the following command:
rosrun ORB_SLAM3 Mono Vocabulary/ORBvoc.txt /home/ORB_SLAM3/Examples/ROS/ORB_SLAM3/Asus.yaml
error：
[rosrun] Couldn't find executable named Mono below /home/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3
There is no executable program Mono in my code. Whether using build_ros.sh or build.sh, it compiles successfully.
Executing roslaunch usb_cam usb_cam-test.launch functions are also normal;

Using cmake to compile Mono in the /home/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3# directory will produce an error
/home/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3/src/ros_mono_inertial.cc:28:9: fatal error: ros/ros.h: No such file or directory
   28 | #include<ros/ros.h>
      |         ^~~~~~~~~~~
compilation terminated.
And the following files cannot be found：
#include<ros/ros.h>
#include <cv_bridge/cv_bridge.h>
#include<opencv2/core/core.hpp>
#include"../../../include/System.h"

My ENV：
ROS_VERSION=1
ROS_PYTHON_VERSION=3
ROS_PACKAGE_PATH=/root/catkin_ws/src:/opt/ros/noetic/share:/home/ORB_SLAM3/Examples_old/ROS/ORB_SLAM3
ROSLISP_PACKAGE_DIRECTORIES=/root/catkin_ws/devel/share/common-lisp
ROS_ETC_DIR=/opt/ros/noetic/etc/ros
ROS_MASTER_URI=http://localhost:11311
ROS_ROOT=/opt/ros/noetic/share/ros
ROS_DISTRO=noetic

Version used：
PRETTY_NAME="Ubuntu 20.04.6 LTS"

I need your help！！！！

soumya997, 2024-11-23:
dude, could you find a solution to this?

soumya997, 2024-11-24:
Okay, I found the solution. 

You need to change the `cd [path]` in `build_ros.sh` file. 

that file has `cd Example/ROS/`, which doesn't exists, change it to `cd Example_old/ROS/`