UZ-SLAMLab/ORB_SLAM3 #959: How can I make ORB-SLAM work on macOS?

tanshiv2, 2025-02-13:


BALA22-cyber, 2025-02-19:
You can try using robostack for setting up ROS on macOS.
Try following their documentation

https://robostack.github.io/#why-robostack

tanshiv2, 2025-02-20:
Hi, thanks for the input. I solved this problem by running dockerized ORB-slam3 from this particular PR because it had the necessary changes for it to work on M1 Macs.

https://github.com/jahaniam/orbslam3_docker/pull/13

Followed the instructions here and after before executing build.sh from orb-slam3 main repo. I had to change lines 7, 16, 31 to `make -j4` instead `of make -j `(this uses all the available CPU cores and may lead to program crash during cmake operations)

closed 2025-02-20 by tanshiv2

Dmitrii173173, 2026-03-02:
https://github.com/Dmitrii173173/ORB-SLAM3-macOS-arm64/