UZ-SLAMLab/ORB_SLAM3 #961: Problems running the stereo-inertial mode...

SrOdranoel, 2025-02-24:
Hello everyone,

I’m currently working with ORB-SLAM 3 as part of my research, specifically focusing on the stereo-inertial mode. However, I’ve been facing some challenges that I hope someone in the community might be able to help me with.
Context:
I couldn’t find any ORB-SLAM 3 repositories compatible with ROS 2, so I decided to build a Docker container using an Ubuntu 20 image to run ORB-SLAM 3.
After resolving several initial issues, I’m now encountering a problem related to performance or calibration.

Error Log:

```
First KF:4; Map init KF:3  
New Map created with 19 points  
TRACK_REF_KF: Less than 15 matches!!  
Fail to track local map!  
LM: Active map reset received  
LM: Active map reset, waiting...  
LM: Resetting current map in Local Mapping...  
LM: End resetting Local Mapping...  
LM: Reset free the mutex  
LM: Active map reset, Done!!!  
mnFirstFrameId = 117  
mnInitialFrameId = 40  
5 Frames set to lost  
not IMU meas  
First KF:5; Map init KF:4  
New Map created with 19 points  
TRACK_REF_KF: Less than 15 matches!!  
Fail to track local map!  
LM: Active map reset received  
LM: Active map reset, waiting...  
LM: Resetting current map in Local Mapping...  
LM: End resetting Local Mapping...  
LM: Reset free the mutex  
LM: Active map reset, Done!!!  
mnFirstFrameId = 120  
mnInitialFrameId = 118  
6 Frames set to lost  
not IMU meas  
not enough acceleration  
not enough acceleration  
not enough acceleration  
```

The following error keeps occurring during execution:

What I’ve Tried:
 I’ve already checked the calibration of my stereo camera and IMU, and they seem to be correct.

I’ve also ensured that the Docker environment is properly configured and that all dependencies are installed.

Questions:

Has anyone encountered a similar issue with the stereo-inertial mode in ORB-SLAM 3?

If so, how did you resolve it? Are there specific calibration steps or parameters I should double-check?

Could the issue be related to the IMU data or the way it’s being processed?

Any insights, suggestions, or references to relevant documentation would be greatly appreciated! Thank you in advance for your time and help.

VASISHTAKODUMAGULLA, 2025-02-25:
Hello @SrOdranoel , did you face the issue with Pangolin? since the display option is not possible in Docker environments I faced some issue with Pangolin docker environment or did you disable the Pangolin?

VASISHTAKODUMAGULLA, 2025-02-25:
Well in my case I tried in Virtual box with stereo inertial mode and I faced the same issue similar to6 Frames set to lost  
not IMU meas  
not enough acceleration  
not enough acceleration  
not enough acceleration  and I heard that is localization issue but I apparently don't have a solution to this yet please let me know if you find an answer to this issue

SrOdranoel, 2025-02-26:
Hello Mr @VASISHTAKODUMAGULLA  about your fist message i had this problem but i solve just giving permission for the docker to acess  and open the display:
```
docker run -it \
    --rm \
    --name orbslam3 \
    -e "DISPLAY=$DISPLAY" \
    -e "QT_X11_NO_MITSHM=1" \
    -v "/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    -v "/etc/localtime:/etc/localtime:ro" \
    -v "/dev/input:/dev/input" \
    -e "XAUTHORITY=$XAUTH" \
    -e ROS_IP=127.0.0.1 \
    --privileged --net=host --ipc=host \
    --cap-add=SYS_PTRACE \
    -v `pwd`/Datasets:/Datasets \
    -v /etc/group:/etc/group:ro \
    -v /dev:/dev \
    -v `pwd`/spinnaker-2.6.0.160-amd64:/spinnaker-2.6.0.160-amd64 \
    orbslam3:latest
```
this is all permission i had give. I Dont know if you want just test the examples but if you are in the situation like me using your cameras and IMU you will need give permission on the docker about the USB port too.

Now about your second message i dont find any solution yet. I send too you a invite on linkedin. we can talk better there and can try too find any solution.

soysaucewaso, 2025-04-16:
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/435

I was facing the same issues. It looks like IMU messes orbslam3 up for cars.
With stereo-only, I'm only able to map rooms which are full of features